EVA-3D/eva-main #1: General: Missing screw holes

miklschmidt, 2020-11-22:
I've noticed that some screw holes are missing in a couple of parts. The few I've noticed so far: The V6 shroud, core xy back (3 holes for universal cable mount), the universal cable mount itself is missing from the EVA 2 repositories, but is missing holes in newest EVA 1 as well. I will keep an eye out as i begin to print the EVA 2 parts and list them here.

pkucmus, 2020-11-23:
If you are talking about for e.g. those: 
[image]
then that's my ~time saving~ professional technique that allows the holes to be printed mid-air. One then needs to poke or drill the holes before assembly.

miklschmidt, 2020-11-23:
@pkucmus ooooh. I see! Fair enough. I might've missed the memo on that one.

pkucmus, 2020-11-23:
Still a valid note to self to document that. Thanks!

miklschmidt, 2020-11-23:
@pkucmus i've updated the issue to reflect the needed change :)

pkucmus, 2020-11-30:
Fixed: https://eva-3d.github.io/eva-main/printing/#pokey-pokey-hole-technique

closed 2020-11-30 by pkucmus