EVA-3D/eva-main #3: Feature Request: Alternative blower fan sizes.

miklschmidt, 2020-12-07:
It would be really cool if we could swap the 90 deg back for a 5020 or even 6015 version. I don't know if larger ones will be possible but... basically as large as we can go. Here's a few options:
[image]

miklschmidt, 2020-12-07:
There's like a 30% difference between the 5015 and 6015 sunons in airflow. Moving to a 7030 would give us almost 3,5 times the airflow at the same noise level. But that one is a chonker (would love it though :D)!

pkucmus, 2021-05-09:
@miklschmidt anything to share / document here from the work done on Discord?

miklschmidt, 2021-05-09:
@pkucmus Yes!

7530 seems to outperform dual 5015's in the test i've done so far. Which makes sense because it's as stated previously quite a lot more airflow. The duct in it's current form does not benefit much from a 7530 or dual 5015's as there's simply not enough room to move more air. What we had to do is increase the width of the bottom duct by 10mm, and beef up the horns to something like 2x, and add a middle duct on top of that. Regarding fans, there's a really really interesting option that i'm trying to get my hands on, the Delta BCB0812UHN, a 75x80x30 fan. It has 5 times the pressure of my current Sunon 5015's and 10 (!!) times the airflow. Still waiting to receive one.

Supersniffles came up with a pretty elegant Dual 5015 fan design that only costs around 2mm on Y on the 1.3, no space is lost on the V3 (except if you have stuff mounted on the back extrusion, unfortunately). But it looks like the ducts aren't quite wide enough to fully utilize the airflow of the 5015s.

miklschmidt, 2021-05-09:
With the 7530, the 10mm wider duct and the "monster horns duct" from supersniffles i succesfully printed a 20 minute benchy without any signs of cooling problems.

miklschmidt, 2021-05-09:
I don't know if it makes sense to look at something like the HevACS instead, but it's just such a brute force solution 😂

miklschmidt, 2021-05-09:
One thing is for sure though, the current duct design and single 5015 is underpowered relative to the performance we can get out of the Rat Rig machines currently. That's a shame. It's very hard to print a benchy faster than 50 minutes with the current design. You can't print 45 degree overhangs much faster than 50-70 mm/s (depending on the fan). It's just not good enough, given the hardware we have available, so some solution needs to be found, in my opinion.

flow23, 2021-05-12:
Oh yeah. Bigger fans. This would also help the V-Cast I guess.

pkucmus, 2021-05-15:
Should we change the standard and increase the fan output on all `back` part?

miklschmidt, 2021-05-15:
Unless there’s a good reason not to, i would say yes!

closed 2021-09-04 by stale[bot]