EVA-3D/eva-main #8: EVA on BLV

pkucmus, 2020-12-31:
Moved from EVA 1: https://github.com/pkucmus/EVA/issues/32

pkucmus, 2020-12-31:
This maybe a more of a BLV project than EVA project but for the sake of tracking I'll leave this here

morpheus200, 2021-03-03:
After your upgrade to EVA 2 and support of the Mosquito Hotend (Thanks a lot) i have printed all parts and have also encountered this problem. Mirroring the front helps but for the rear side this is not possible because the fan is the wrong way round.
It looks like it is a problem with the BLV Cube design, because the belts are guided differently.

damik422, 2021-03-08:
Hey I have a hevort not a BLV but the belt path is similar. I have a problem, the rear belts are flipped so that the teeth side faces away from the 2020 extrusion. The belt holders are designed for the belts facing the other way around. Is there a way to mitigate that or do i have to modify the belt holders so that they have teeth on the other side of the channel for the belt? ps. im using 6mm belts

pkucmus, 2021-03-09:
> Hey I have a hevort not a BLV but the belt path is similar. I have a problem, the rear belts are flipped so that the teeth side faces away from the 2020 extrusion. The belt holders are designed for [...]

Please check in the Onshape document in the 'backs' folder. There's a high chance those are already there.

pkucmus, 2021-03-09:
Here: https://cad.onshape.com/documents/1765b04fac582f6c1c470bd3/w/1cc31596374d6ce51cd23fa9/e/626c5e88e2db7d8a29b52142
[image]

tetele, 2021-03-10:
Finally published the conversion kit for the BLV MGN Cube on the contributions repo https://contrib.eva-3d.page/printer-compatibility/blv-mgn-cube/

closed 2021-09-04 by stale[bot]