EVA-3D/eva-main #9: Duet Mini IR sensor

pkucmus, 2020-12-31:
Moved from EVA 1: https://github.com/pkucmus/EVA/issues/18

pkucmus, 2021-01-08:
[image]
[image]

Few things I was trying to consider:
1. how the pins are aligned, they go outwards into X-, I made a cut to loop the wires under the probe mount and guide back
2. since I haven't tested it yet and never used it before I don't know if the cover makes sense or not
3. I initially tried a different orientation with the electronics facing the hotend but that did not feel comfortable with that, especially when Mosquito is considered (wide heat block)

tetele, 2021-01-10:
@pkucmus the capacitor is not necessarily that short in all models, so in my case, i the cap would be unusable because the capacitor would have nowhere to go https://imgur.com/a/9IULosT

Also, keep in mind that the usable range of the IR sensor is pretty small, so there would have to be brackets for different height hotends (I'm thinking about the Volcano in particular here).

Ronniehertog, 2021-01-30:
Hi all, 
 
 I was looking for a IR probe mount, found this and 're-desinged' it so it can slide up and down, and to be firmly attached using the embeded nuts on the left hand side.

[IR sensor mount v2.zip](https://github.com/EVA-3D/eva-main/files/5898535/IR.sensor.mount.v2.zip)

tetele, 2021-02-01:
@Ronniehertog some "default position" indicators for V6 and Volcano standard height screw positions would make it great. But that's just nitpicking.

Apart from that, I have one single issue with this design: not everybody will be able to print the long bridge in the regular printing position (i.e. right hand side of the screw channel).

But a very elegant solution, congrats!