EVA-3D/eva-main #10: EVA 2 / Orbiter (Orbiter support)

pkucmus, 2021-01-01:
[image]

MGN 12: https://cad.onshape.com/documents/bb8303124bc8370b6d1924bc/w/8276a22c579374e8b02e9b96/e/6b8206d26df549409bc1d1b2

MGN 15: https://cad.onshape.com/documents/551442bc5664b4216eddf506/w/16749997681c8bec66757f15/e/8952ad2027c63b84e64fdcba

pkucmus, 2021-01-16:
Added the Orbiter CR-10 version to both MGN12 and MGN15

tetele, 2021-01-19:
Using this on a BLV MGN Cube with a Volcano 0.4mm nozzle. The assembly went great, as with all EVA components.

Here's a sample print at 0.2mm height, 80mm/s perimeter speed and 100mm/s infill speed. Acceleration 3000, except for external perimeter, which is capped at 300 for a better finish. Colorfabb economy PLA black @ 200deg

I could have gone higher with the acceleration, but I haven't really tuned it yet.

https://imgur.com/a/QLx88BT

[image]

[image]

closed 2021-02-04 by pkucmus