EVA-3D/eva-main #11: BLTouch mount interferes with probe

shuffleznl, 2021-01-07:
First of all, many thanks for a great design. Amazing work!

Quick one on the BLtouch Standard mount; the vertical overhang-reinforcement to support the horizontal BLtouch mount area interferes with the mounting 'flange' of the BLtouch itself.

It's only ever so slight, may just be 1mm off in the design.
[image]

pkucmus, 2021-01-07:
Thank you for the ticket, I'll look into it

pkucmus, 2021-01-07:
Looking at the picture, there's a chance your printer is overextruding a bit but the lack of tolerance on the design is an issue I just fixed.
You can find the new part here:
https://cad.onshape.com/documents/94cafeab52e3dfa0a28bae89/w/99794094315556b24b5664ec/e/d6374a1e5b14711360d3b164

[image]

Please let me know if it works.

shuffleznl, 2021-01-07:
Extrusion still needs to be tuned yes :) Printed way too fast (💯mm/s) on 0.15 with very poor non-dried 'transparent' Amazon filament :). Currently still run V-Core (rails) w/o EasyMod and EVA, printed EVA components fit like a glove though, your design is awesome! Will print all again slower with Easy & EVA ❤️ 

Printing now.

closed 2021-01-07 by shuffleznl