EVA-3D/eva-main #16: Published STL's of the Bowden top does not fit MGN12H

rveda, 2021-01-13:
## Describe the bug
Published STL's of the Bowden top does not fit MGN12H

## Actual behavior
The hole pattern matches MGN12C

## Expected behavior
The STL file to have the enlogated screw holes like in the CAD model.

## Pictures
I exported the top piece from the CAD model and compared it to the STL
[image]

## Device
Hemera download and the complete download

rveda, 2021-01-13:
Sorry. Didn't realize it was not released yet. You can close this Issue.

pkucmus, 2021-01-14:
For future reference, the parts in question are yet to be released and I need to do a better job describing that the Onshape links are always bleeding edge.

closed 2021-01-14 by pkucmus