EVA-3D/eva-main #17: [EP] Support larger motors on the BMG direct drive

tetele, 2021-01-14:
## Problem description
If you use a motor that's larger than the 25mm pancake in the design, it would hit the universal cable mount and not fit

## Describe the solution you'd like
I'd like to have either a universal cable mount or cable holder mounting point on the top BMG direct drive plate.

### Functional and non-functional requirements
- have the possibility to use larger motors for the direct drive BMG and also have a cable holder
- have alternative mount points for the cable holder

### Proposed solution
Widen the top hole on the universal cable mount (the one that's not countersunk for bolt heads) to 3mm and align it with the side holes on the top piece so that you can mount the cable holder laterally. https://www.screencast.com/t/a7jK3fgvZad

### Considered Alternatives
Only bad ideas :D

### Additional context
The BLV MGN Cube BOM includes a 40mm extruder motor that doesn't fit the EVA 2.1.1 mounts.

pkucmus, 2021-01-14:
I have a few comments (open for discussion):
1. Use 25mm or 23mm NEMAs - if someone has access to those then those should be used to reduce the X gantry weight, I understand that not everybody wants to pony up for one hence more points...
2. This could be an adapter to move the cable holder to the side [image] 
3. Probably the least involved idea, move the holes on the cable holder part to match the extra side holes on the BMG and Titan parts (that's what those were for initially): [image]

EduardoOliveira, 2021-01-14:
>    2. This could be an adapter to move the cable holder to the side [image:
>    image]

As a quick and dirty solution, once upon a time I inserted m3 nuts as
spacers in between the carriage and the cable holder to accommodate the e3d
pancake wich is bigger than normal, I don't believe it to be 40mm thought.

pkucmus, 2021-01-14:
At one point only 23mm pancakes fit, now 25mm (like the E3D ones) are supported as well

tetele, 2021-01-14:
@pkucmus regarding point 3 above, those holes don't line up. I think it would be a good enough idea if it were feasible, but it's not. What should change? The top or the cable holder?

tetele, 2021-01-15:
That would imply redesigning the entire holder, as you can't have the holes poking through the channels meant for zip ties. Also, you'd get less wall thickness if you move those holes to be 10mm apart instead of the 7.4mm that's currently between the "fingers" of the cable holder.

So my question is: why not enlarge the top hole on the universal cable mount to 3mm diameter and align it so that it's exactly 10mm apart (center to center) from the bottom screw hole on the same piece? That would fit nicely in the sidewall of the direct drive top mount.

pkucmus, 2021-01-15:
Those extra holes are to grab a thread of filament to make the cable bunch stiffer (like the nylon trick Prusa does). I just think there are more solutions to that problem :) To narrow down and add one other:
1. Making a new cable holder part - I still think it's possible, just the ziptie spacing would need to change
2. Do what you are describing and make a different cable post that would take the 2 hole 10mm spacing on the side.

The tops are complex (in maintenance) and should be last to change.

tetele, 2021-01-18:
@pkucmus any favorites?

[image]
[image]

pkucmus, 2021-01-18:
I like them both. From experience I can tell you that 2 zip ties is sometimes not enough so the first one my behave better.

tetele, 2021-01-18:
My thoughts exactly. I'll edit the universal cable mount post and upload them both here. Thanks!

tetele, 2021-01-18:
[image]

[modified cable holder.zip](https://github.com/EVA-3D/eva-main/files/5829766/modified.cable.holder.zip)

closed 2021-09-04 by stale[bot]