EVA-3D/eva-main #19: [EP] Support for Bondtech LGX

pkucmus, 2021-01-17:
## Problem description
Bondtech is releasing this new (face heavy) extruder/hotend which is already of high demand. EVA should support it especially that @ogland already made the work for EVA 1.

## Describe the solution you'd like

General support for Bondtech LGX

### Functional and non-functional requirements
- as a first pass, port @ogland's work to EVA 2

### Proposed solution

None

### Considered Alternatives

None

### Additional context

None

pkucmus, 2021-03-27:
Open for testing: https://cad.onshape.com/documents/d2a2f19b34851a8570487add/w/1ff9bb9662c715a9cdc3d984/e/f74cb56cf915933a84ff4023
[image]

ogland, 2021-03-27:
Looks good! Looking forward to seeing your approach to LGX FF.

pkucmus, 2021-03-27:
@ogland I aim to port yours as well and release both

pkucmus, 2021-03-27:
Would be cool if it was printable... don't print - there's still much work to do with this on.

pkucmus, 2021-04-25:
Deployed in 2.3.0: https://main.eva-3d.page/drives/lgx/

closed 2021-04-25 by pkucmus