EVA-3D/eva-main #21: [EP] Support for 4020 extrusions.

simonjohansson, 2021-01-25:
## Problem description
I have a Sovol SV01 which comes with a 4020 extrusion for the X axis. The current design assumes a 2020 extrusion

## Describe the solution you'd like

Adapt the back and front plate for 4020 extrusions.

### Functional and non-functional requirements
- functional requirement
- functional requirement
- non-functional requirement

### Proposed solution

I just glanced over the CAD drawings and I think if the front and back plate would just extend 20mm at the top everything would work with a longer boden tube between the extruder and the hotend?

### Considered Alternatives
Put a 2020 extrusion on my printer, but I don't want to faff around with drilling and tapping holes :)

liner238, 2021-02-27:
Hello! , please raise this topic again. A similar problem. there is a 20x40 profile and I can't put EVA on it.
Because of the import curve in solidworks, I can't do it myself.

liner238, 2021-02-27:
while I'm waiting for the mod, I tried to do it myself. It turned out scary, but it should work. I will add it to print.

https://drive.google.com/file/d/1kOXiTMUfa8gKGMt0tax1G64r5P_7PTy_/view?usp=sharing

pkucmus, 2021-03-05:
EVA is a MGN 2020 based platform - a defined standard a printer needs to follow. So it's more about a printer working with EVA than EVA working with a printer. I understand the need but that will simply not be EVA but maybe a EVA-4020 fork.

I hope you'll understand.

closed 2021-03-05 by pkucmus