EVA-3D/eva-main #27: [EP]

SmallGreenParrot, 2021-03-01:
## Problem description
The Mosquito - Bondtech scenario looks good.
But with (a) the extruder to hot-end filament path hidden, and (b) the heater / thermistor oriented to the front, early problem discovery is difficult.

## Describe the solution you'd like
(a) Exposing the filament path between the hot-end and extruder would allow hands-free inspection. It would also avoid an issue like having a missing PTFE tube.
(b) If the Mosquito was turned 180 degrees with the heater at the back,  it would make the nozzle more easily visible.
Doing that makes the fan shroud design more of a challenge but the end result would assist  in fault discovery.

### Functional and non-functional requirements
- visibility of the extruder to hot-end filament path. 
- better visibility of the mosquito nozzle.
- careful repositioning of the fan air shroud path. to avoid heat issues.

### Proposed solution

### Considered Alternatives

### Additional context

pkucmus, 2021-03-05:
I'm not sure I'm following:
re a) what inspection are you thinking about here? what is the use case? missing PTFE tube - that part I completely don't understand, if it's missing it's an assembly error on the user, right?
re b) This is a "wont do" for 2 reasons - there is no space for that and the support part and I don't want that heat to be closed there

Please let me know if I got something wrong

SmallGreenParrot, 2021-03-05:
When everything is working fine, visibility into the filament path is 
just for interest sake.

When there is a problem, that is where visibility and easy access become 
important.

Not focusing on problems with the design,
I was trying provide a wish-list of features in an effort to contribute 
to the creation process.

For example, the Micro Swiss line of extruders have excellent visibility 
without touching anything.
https://store.micro-swiss.com/collections/extruders

I have had some recent frustration servicing a different 3D printer and 
I thought you might appreciate some positive suggestions.
Having just ordered a kit from RatRig, I will soon face these issues.

My thought processes are just;
"If there was a problem with the filament somewhere, how would I go 
about finding the problem and fixing it?"
"Is it possible to improve the design to assist in that?"

I have even thought further about how I might go changing the ho-end for 
a different brand?
Would it have to include disassembly of the drive belt mechanism to do that?
The Prusa MK3 is like that. You have to take everything apart, making it 
a really big job.

The many hours of work that have gone into the designs is obvious.
I hope my attempt at constructive contribution have not been taken 
negatively.

Regards,
Noel

pkucmus, 2021-03-24:
Sorry, I missed your reply somehow. My English maybe failing me here as I'm still not sure if I'm following but we maybe slightly invading the "over-engineering" territory here - which by design principle I'm am strictly avoiding.
I am grateful for all feedback (sorry if I sounded mean). 
Discord might be a better place for a quick discussion about this, would you care to reach me out there?

pkucmus, 2021-03-29:
Closing this one, but feel free to comment if need be

closed 2021-03-29 by pkucmus