EVA-3D/eva-main #28: [EP] Copperhead

pkucmus, 2021-03-01:
## Problem description

I'd like to add Copperhead to the list of supported hotends.

## Describe the solution you'd like

Copperhead needs to be a fully supported hotend in the standard

### Functional and non-functional requirements
- Needs to work with the universal face
- Tip of the nozzle needs to be in the same point in space relative to the carriage as V6/Mosquito/Dragon
- No groovemount
- Support 40mm hotend fan

### Proposed solution

Something along those lines: 
[image]

### Considered Alternatives
N/A

### Additional context
I'm waiting for the delivery of one I got from donations (thanks).

pkucmus, 2021-03-27:
Open for testing: https://cad.onshape.com/documents/980a8bb4514542eeb90c5c21/w/fd1e2bf7da865e46d00cc945/e/c8b435b40390c0eb453f4c9e
[image]

pkucmus, 2021-04-25:
Deployed in 2.3.0: https://main.eva-3d.page/hotends/Copperhead/

closed 2021-04-25 by pkucmus

benrr101, 2023-11-23:
It seems the copperhead was dropped in v3. Is that correct? Are there any plans to bring it back in a later release?