EVA-3D/eva-main #29: [EP] Orbiter 1.5 (and 2.0?) support

pkucmus, 2021-03-01:
## Problem description
A few days after EVA / Orbiter was released The Orbiter 1.5 was released I need to figure out support for 1.0, 1.5 and possibly 2.0 is it gets here before I will be able to address this.

## Describe the solution you'd like

Make an additional variants of EVA / Orbiter for different Orbiter versions

### Functional and non-functional requirements
- Default v1 Orbiter support has to remain
- CR-10 v1 Orbiter support has to remain
- Orbiter v1.5 support is added
- Orbiter v2.0 support is added (if possible)

### Proposed solution

Not sure yet, but possibly it's just a different mounting screw alignment for 1.5 to fit on EVA

### Considered Alternatives
N/A

### Additional context
N/A

Oranginator, 2021-03-02:
Here you go, this is for Orbiter 1.5 standard. I simply hat to move the mounting holes a bit.
[Orbiter 15 mount.zip](https://github.com/EVA-3D/eva-main/files/6069406/Orbiter.15.mount.zip)

pkucmus, 2021-03-06:
Deprecating Orbiter 1.0 support, it will be available in git history (I'll link in the documentation). 
Orbiter 1.5 support: https://cad.onshape.com/documents/bb8303124bc8370b6d1924bc/w/8276a22c579374e8b02e9b96/e/6b8206d26df549409bc1d1b2

closed 2021-03-06 by pkucmus

pranas, 2021-07-17:
STLs are available in history but is there a way to get history from Onshape (the CAD file for particular piece)?

pranas, 2021-07-19:
It's here:  https://cad.onshape.com/documents/bb8303124bc8370b6d1924bc/v/aef34c7fede34b1b25331696/e/09d4e72250a20923ab87fce6

PR #30 (2021-03-06): Release/2.2.1
resolves #26
resolves #29 
resolves #24 
resolves #22