EVA-3D/eva-main #32: Nut clearance at join of "universal face" and "bottom_mgn12_short_duct"

pman92, 2021-03-20:
The captive nut hex shape at the bottom of "universal_face.stl" is too short for a M3 nut to be fully pulled in flush. It sticks out and prevents "bottom_mgn12_short_duct.stl" from mating properly. This nut is used to secure "v6_support.stl".

There is room in "bottom_mgn12_short_duct.stl" to provide a clearance hole so the bolt and nut could both protrude into them. 

[image](https://imgur.com/a/jIgDy7Z)

silvalis, 2021-03-20:
As per the duct alignment issue, you've printer the incorrect bottom_mgn12_short_duct. The one in the orbiter fileset has a cutout to allow clearance for the nut and screw.

pkucmus, 2021-03-20:
There might be an issue with the stl directory in the root of the project, in the image bellow you'll find a preview of the proper file: 
[image]
https://cad.onshape.com/documents/1765b04fac582f6c1c470bd3/w/1cc31596374d6ce51cd23fa9/e/bf748cd402ac505d88311d6e

pkucmus, 2021-03-20:
It was a filename issue it's fixed in: https://github.com/EVA-3D/eva-main/commit/0f7b9bcc789e196c75741722d1397ba23493656d
Thanks

closed 2021-03-20 by pkucmus