EVA-3D/eva-main #33: Part cooling fan duct alignment with hot end (MGN12, cartesian, dragon hotend)

pman92, 2021-03-20:
This may be an issue with me using the wrong stl's or not really an issue at all (I'm unsure).
I'm printing parts to build a cartesian printer using MGN12, an orbiter extruder and dragon hot-end. I'm still waiting on the hotend to arrive, but have noticed after printing that the output of the fan duct is sitting far forward of the center of the filament path. 
I'm using "universal face" and "dragon_face" at the front, "back_cartesian" at the back, and "bottom_mgn12_short_duct" with "Duct V2.1 straight" at the bottom (as listed at https://main.eva-3d.page/drives/orbiter/)
It almost appears that without the spacer that is part of "bottom_mgn12_short_duct" pushing the duct further forward it would be in the right spot (IE. if "Duct V2.1 straight" was bolted directly to "back_cartesian")

[image 1](https://imgur.com/0tkUchp)

[image 2](https://imgur.com/undefined)

silvalis, 2021-03-20:
Looks like you've printed the wrong bottom_mgn12_short_duct. Maybe the one from the Aero or the Hemera fileset. Use the one in your orbiter link above.

pkucmus, 2021-03-20:
It was a filename issue it's fixed in: https://github.com/EVA-3D/eva-main/commit/0f7b9bcc789e196c75741722d1397ba23493656d

Please print https://github.com/EVA-3D/eva-main/blob/main/stls/bottom_mgn12_short_duct.stl

Thanks

closed 2021-03-20 by pkucmus