EVA-3D/eva-main #35: [EP] Compability with piezo orion probe.

weawer, 2021-03-23:
## Problem description
Start using the EVA system with an e3d v6 and an piezo orion sensor

## Describe the solution you'd like

For the EVA system to have a modified hotend holder and fan ducts to accommodate the extra size this sensor needs for the PCB and holder.

### Proposed solution

The module for the mounts builds about 19mm to the height and is 35mm wide. It is placed between the mount and the hotend. 
This means that the fan ducts would need to be about 19mm lower and a bit longer. I believe the face assembly would be the biggest hurdle since it would have to both be elongated and modified to make room for the piezo mount and pcb/wires. 

Step/stl files for pcb and mount for the piezo orion can be found here
https://www.thingiverse.com/thing:2841681/files

pkucmus, 2021-03-24:
This topic is returning from time to time. Sadly it's not something that I'm able or willing to do as it would require different set of parts for all hotends - not something I can maintain in the long run.
I also want to keep the tip of the nozzle in Z as a part of the spec - in short Piezo Orion does not fit EVA, but I'm open to discussion

weawer, 2021-03-24:
@pkucmus thanks for your quick response. i understand the issue. Thank you for the consideration.

closed 2021-03-25 by pkucmus

pawlowskialex, 2021-06-17:
Sorry to resurrect an old thread, but I've been thinking about a similar thing, albeit with a different spin.
Instead trying to fit the whole Piezo Orion, we can take a small piezo transducer, and put it behind the hot end directly, something like this.

[image]
https://cad.onshape.com/documents/9b313e797dbfe82298fa97a9/w/bfab425eae9c42bf816e1e9c/e/a83baf260163688de249ca06

The extrusion I drew is way to small to host a real transducer, but there are 12 mm transducers out there which should fit.
We'd need to provide a small amount of flexibility for the transducer to register pressure, and this can be accomplished through a shallow counterbore on the hotend mount, where we will put a short spring.

[image]
https://cad.onshape.com/documents/9b313e797dbfe82298fa97a9/w/bfab425eae9c42bf816e1e9c/e/b189a406e6de5fcfe5c86762

Then, when we have mechanics sorted out, we can use the amazing [Pyro-Piezo](https://docs.pyroballpcbs.com/) PCB to drive the whole thing.

(Pardon my Onshape skills)

pkucmus, 2021-06-17:
@pawlowskialex this is one of the best issue comments I ever got, thank you for that.
I am struggling a little here as we put so much work into making the hotend as stiff as possible and then piezo options need some slack. Maybe someone needs to teach me more about it but right now I just don't like the whole piezo option. Additionally I'd like to learn what is the benefit - it's not a more accessible / easier option and the others - Bltouch and Inductive probes seem to be more than enough accurate.
Again awesome work preparing this, we need more comments like that but piezo is just not a fit here or someone needs to help me to stop being ignorant about it :)

pawlowskialex, 2021-06-17:
I can totally understand your hesitation and I’m not 100% convinced this is a good idea neither. It will indeed add some mechanical instability.
But, the reason I think this might be interesting is the potential benefit to non-planar printing it might bring. Having a BLTouch attached to your extruder really limits the range of safe motion and introduces a risk of your probe ramming into the print.
And to throw ideas out there, there are other potential mounting options for the piezo than the hotend: either side of the gantry, on the rail slide, or even under the bed (but it’s probably not the best option for V-Core 3)
But this really requires experimenting and mechanical expertise (which I don’t have yet)

pawlowskialex, 2021-06-18:
Another interesting solution that doesn't require introducing flex into the frame: https://github.com/KevinOConnor/klipper/issues/3741