EVA-3D/eva-main #38: [EP] Add Support For The HextrudORT Extruder

sirhavoc, 2021-04-01:
A nice and lightweight 215g all in one extruder and hot end combo.

https://github.com/MirageC79/HextrudORT

themacboy, 2021-04-11:
Interesting option ! Currently I doubt about orbital extruder, sherpa or HextrudORT. 
That last turns more interesting if I can use on EVA ;)

pkucmus, 2021-04-11:
There is one already: https://www.youtube.com/watch?v=x7njfPCIbXY I asked him if he is willing to share his mod

simonvez, 2021-04-11:
a nicer video https://www.youtube.com/watch?v=6JBIEJan6zs&t=273s 

I can share. Do you need MGN12 or MGN9 ?

pkucmus, 2021-04-11:
MGN12 please, I'll make the MGN15 out of that and we can make it an official thing

simonvez, 2021-04-11:
MGN15 is too heavy.... well MGN12 is too heavy as well lol... but here you go
https://1drv.ms/u/s!AkTq2jZffWF0gocHkFKHoZjcX35RRw?e=hdV0Km

pkucmus, 2022-07-05:
I thought it might go into 3.1 but it's more difficult than I expected. Need to take it down from 3.1.