EVA-3D/eva-main #40: [EP]Eva2 Front for the new Phaetus DragonFly HIC HF hotend

cyborgcnc, 2021-04-05:
## Problem description
Phaetus has just introduced their new hotend, DragonFly HIC HF.  This seems to have a lot of promise, as it has the
heartbreak AND the Nozzle Bonded/Welded together!  This should basically eliminate the possibility of ANY leaks, and
improperly tightened nozzles in the 3d printing world.  It is a high flow device, which when combined with a .6 or .8 nozzle, should really make a coreXY push a lot of plastic (needs to be proven)  More information can be found here: 

https://www.phaetus.com/?page_id=1482

I would like EVA2 if possible to support this HotEnd via a "front"

## Describe the solution you'd like

An EVA front end, that would be able to accept this hotend.  I guess all the other options for extruders, should remain the same.
This should also be a nice fit for the new BMG LGX, and should also pair nicely with an orbiter.

themacboy, 2021-04-11:
Dragonfly BMO in not compatible with Dragon socket?

pkucmus, 2021-04-11:
In theory this should work either without the groove mount with the dragon face since the hole spacing is the same: 
[image]
[image]

or with the V6 face and the groove mount, I don't have this hotend so someone would need to test it

themacboy, 2021-04-13:
Confirmed, they are compatible with dragon

[image]

themacboy, 2021-04-18:
Oh sorry I confuser Phaetus Dragon Fly HIC HF with the BMO. 

Im sorry, miss understood!

Phaetus Dragon Fly HIC HF is other format:

[image]

3DPrintBeginner, 2021-06-29:
There's a STEP file available for the HIC on [Thingiverse](https://discord.com/channels/582187371529764864/758622323162218496/852110074775732224)

themacboy, 2021-07-01:
I have desing the HIC version. I f you need it please contact me in Discord for betatesting.

pkucmus, 2021-08-28:
Was this ever completed? I can only see the BMS contrib: https://contrib.eva-3d.page/hotends/DragonFly_BMS/

cyborgcnc, 2021-08-28:
Hi Pawel,

Not sure, I was not the one building it or modeling it...

thanks,

Spiro

themacboy, 2021-08-28:
Please check my last PR:

https://github.com/EVA-3D/contrib-extras/pull/28