EVA-3D/eva-main #42: Incorrect files ..

themacboy, 2021-04-24:
In the STS.zip the file "bottom_mgn12_short_duct.stl" looks like a incorrect version.

[image]

Download linl:
https://github.com/EVA-3D/eva-main/raw/2.2.1/stls.zip

File preview:
[image]

But must be:
[image]

Im correct? Thx in advance ! Great project !

pkucmus, 2021-04-25:
I'm about to deploy 2.3.0 today, wrote a lot of code that should do all this automagically so I hope those mistakes will not happen again (taking human error out of the equation)

pkucmus, 2021-04-25:
Deployed, please let me know if something if not right

closed 2021-04-25 by pkucmus