EVA-3D/eva-main #43: Cartesian and Idex [EP]

themacboy, 2021-04-26:
First of all, I'm sorry to post it here, but the Cartesian repository looks close to post issues.

I noticed than cartesian and idex design use the 5015 blower, and it is too large to fit in my Ender 5 Plus, and probalbly others. 

I like to know if is possible to get a blower 4020 version. (never used onshapes, currently learning FreeCad)

Or if it is compatible to use the coreXY version.

Thx in advance ! Great work, soon I will post my pictures.

pkucmus, 2021-04-26:
I'll look for a guy who put EVA on an Ender 5 to he if he can help

pkucmus, 2021-04-26:
Also this is a perfect place for those "EPs" :)

vellkan, 2021-04-26:
> First of all, I'm sorry to post it here, but the Cartesian repository looks close to post issues.
>
> I noticed than cartesian and idex design use the 5015 blower, and it is too large to fit in my [...]

Hi :)
You've got problem with back plate fan, correct? On repository is slim version for 5015 fan, but event with that You need to move endstop.
I've got this in Ender 5 Pro, I just move endstop a little and now everything is fine. I can provide You some photos if You want.

themacboy, 2021-04-26:
Yes, sure. And a lot of Thx

themacboy, 2021-04-26:
Where is the slim version?

vellkan, 2021-04-26:
As I understand You've got this problem, yes?
This is version with slim back plate, but still can hit to this metal parts. I just moved Y endstop to the front.
Yes, I lost about 2cmm of bed in Y axis, but can live with that :)

[image]
[image]
[image]

vellkan, 2021-04-26:
I found it on Discord channel, but I will upload it on OneDrive:
https://1drv.ms/u/s!AglN16Zy9Yb1pmkzyMej11izTrnm?e=GsXgku

themacboy, 2021-04-26:
Lot of Thx

pkucmus, 2021-08-28:
I'll assume this one is closed

closed 2021-08-28 by pkucmus