EVA-3D/eva-main #46: [BUG] Belt indentation inside cartesian back

flow23, 2021-05-11:
## Describe the bug
Belt indentation inside cartesian back.

## Actual behavior
Belt indentation inside cartesian back.

## Expected behavior
No belt indentation inside cartesion back because the belt is secured by two belt blocks in the back.

## Pictures
https://cad.onshape.com/documents/a2d21b0faa4745fa21bc844e/w/d7f14dfeeee2e68fe2f1149b/e/fcd1503d3995b7b7473cc831

## Device
X

## Additional context
X

pkucmus, 2021-05-15:
I am sorry but I will require this ticket to be more descriptive as I have no idea what this is supposed to achieve

flow23, 2021-05-16:
I was wondering if this feature is really needed for cartesian printers? I thought that the belts are tentioned by the two blocks in the back. Maybe I misunderstood the CAD drawing.
[image]

EduardoOliveira, 2021-05-16:
In any configuration each belt has a tip that goes into a tensioning block and another tip that is held statically to the carriage.
In CoreXY systems one of the tips is held by the block in the back and the other tip is held in the front.
In Carterian systems the static tip is held by that groove and the other tip by the block

Having 2 tensioner blocks in a single belt would be redundant at best

Did I make it clearer? If you have any doubt you can  reach out on discord

flow23, 2021-05-16:
Understood.

closed 2021-05-16 by flow23