EVA-3D/eva-main #48: [EP] Sapphire Pro compatible mount

yfaykya, 2021-05-26:
## Problem description
I printed the Orbiter mount but no matter the orientation I lose quite a lot of Y distance due to colliding with the frame.
 

## Describe the solution you'd like

Not sure if possible/easy but perhaps something that can slide over the top carriage.

https://photos.app.goo.gl/uEgFbnfYc6Z9rKTy5

themacboy, 2021-05-26:
Hello ! Im currently near to mount it too.

I few days ago I have modfied CoreXY to fit on a Sapphire Plus, but not sure if it is usefull in the Pro version. 

As soon as posible I will test my mod and report result.

themacboy, 2021-05-26:
This is the mod for the Sapphire Plus:

[image]

As you can see this probably will hit with the rods on the sapphire pro.

But I will try using a 4020 blower or other.

Any idea is welocme

yfaykya, 2021-05-26:
If there is anything I can test please let me know! Appreciate this a lot!

themacboy, 2021-05-26:
how many Y are youcurrently losing ?

yfaykya, 2021-05-26:
50 mm or so.

yfaykya, 2021-06-05:
Can you share what you have and I can try some modifications? Thanks!

themacboy, 2021-06-05:
Currently there is a contributor that has worked on a 4020 fan version for the sapphire. Im currently waiting for him to share the files in the contrib version.

About my work it is not usefull due the belt position in the Sapphire Plus, and still not tried in my sapphire Pro. Im moving to another home, very busy month

themacboy, 2021-06-05:
> 
> 
> Can you share what you have and I can try some modifications? Thanks!

btw, you can use the public OnShape soucer design to make any mod. They are very usefull.

https://cad.onshape.com/documents/1765b04fac582f6c1c470bd3/w/1cc31596374d6ce51cd23fa9/e/e2fc56d355d28a162332ce38

pkucmus, 2021-06-06:
Not fully compatible with EVA 2, but there are still EVA 1 backs available - which had a 30deg back
[image]

Maybe you can go from there: https://cad.onshape.com/documents/1cea28d26150bba34a3d6614/w/fd185c58d38d06bda7ec83a6/e/5385243b86e5aae8a3a0e66d

themacboy, 2021-06-06:
> 
> 
> Not fully compatible with EVA 2, but there are still EVA 1 backs available - which had a 30deg back
> [image]
> 
> Maybe you can go from there: https://cad.onshape.com/documents/1cea28d26150bba34a3d6614/w/fd185c58d38d06bda7ec83a6/e/5385243b86e5aae8a3a0e66d

YEs I was thinking on a similar option, but probably will hit with de back frame belts.

A will try to work in a totally horizintal option. But currently Im noving to other home, and Im full busy. Sorry

themacboy, 2021-06-09:
> @themacboy did you already fit EVA on Sapphire Plus? I'm interested in doing that as well. I saw the [back mod for cartesian](https://main.eva-3d.page/addons/cartesian_idex/) printers which has the blower a lot lower and I hope it could clear below the frame but I didn't take any measurements yet. I would also have to redo belt paths and I wonder maybe you already have that part figured out?

I also tried it on Sapphire plus, and had the same problem, using the Cartesian, with 5015 or 4020, it played with the belt and still forgives a lot of range on the Y axis.

This is from another user (from Discord) with nick XP-ERIMENT, here is the problem:

! [image0] (https://user-images.githubusercontent.com/8063711/121410399-d441eb80-c962-11eb-8c97-5625c1fb116d.jpg)

It looks like he finally modified the seat belt and is working well with it, but he hasn’t shared it in the STL and the images yet to see the modification on the belts.

I will soon try to work with a flat version of the fan. Although I’m not sure if it’s possible to do that.

pranas, 2021-06-10:
Thanks for sharing. I think there's another problem that the stock Sapphire Plus hot end carriage is mounted backwards: filament path in the back. If you mount EVA like it was intended at Y=0 the nozzle would be way too far in the front and out of the bed area. Maybe we should mount EVA backwards as well, that way the cartesian blower would have a lot more space because there is more space in the front due to XY motor placement there. It would not look great but would probably be a better way functionally.

yfaykya, 2021-06-30:
I have not had a chance to check in here for a bit,. Are there any model I can test? Would love to help out.

themacboy, 2021-08-28:
I just tested EVA 2.3 on Sapphire Pro and the bed has been reduced to 220x170 mm and the height is increased to 265 mm

yfaykya, 2021-09-03:
@themacboy Which model did you use?

themacboy, 2021-09-06:
corexy with a few personal modifications in the back plate to better align of belts