EVA-3D/eva-main #49: [FR]

riox87, 2021-06-01:
**Problem description**
The motionsystem on my CoreXY has its belt postions on the opposite: on the front the belt is on the bottom left and top right.

**Describe the solution you'd like**
If possible, a version of back_corexy and universal_face with inverted beltmount postitions.

EduardoOliveira, 2021-06-01:
Hi, have you checked the contrib page (https://contrib.eva-3d.page) ? Maybe there is some adaptation that could solve your problem

riox87, 2021-06-01:
Hello.
Yes i checked it but unfortunately there is no solution for my problem.
Thank your for the quick response.

riox87, 2021-06-01:
I checked it again, you where right.
blv-mgn-cube-back_corexy.stl is what i was looking for.
Thanks a lot!

closed 2021-06-01 by riox87