EVA-3D/eva-main #57: [EP] Support for the Carabiner Toolboard

Kirinkunhime, 2021-08-09:
## Problem description
Wiring the toolhead is very annoying. As such a simple and lightweight toolboard makes wiring and replacing components (e.g. heater, fans) much easier. The 

### Proposed solution
Support on EVA2 for the Carabiner toolboard.

### Considered Alternatives
While the Duet2 toolboard is an option, I personally do not like having screw-terminals to retain a higher voltage wire on a moving part of the printer. 
In addition the Carabiner retains the controlling side of the electronics inside the electronics box and does not have any components inside the printer, which in an enclosed printer (which can reach 50°C chamber temperature) is an additional advantage.

### Additional context
https://github.com/Annex-Engineering/Annex_Engineering_PCBs/tree/master/carabiner-toolboard

Kirinkunhime, 2021-08-09:
Side Note: There are no vendors carrying the Carabiner, it's a completely DIY PCB. If you want one to test, please poke me in Discord ( iKirin#0001 ) and I can provide you one 👍

alecwarren, 2021-08-12:
> Side Note: There are no vendors carrying the Carabiner, it's a completely DIY PCB. If you want one to test, please poke me in Discord ( iKirin#1 ) and I can provide you one 👍

I like the design and size of that toolboard. Too bad it only has a 4 pin probe port.

Kirinkunhime, 2021-08-16:
Thank you!

I might be doing a revision that will have 5 pins for the probe-port with the 5th port being connected to one of the other 4 - basically allowing you to drop in a BLTouch without any major hassle.

Kirinkunhime, 2021-09-15:
This ping is to indicate further interest in a mount for the EVA2.

wzzx, 2022-03-30:
Would love to see this coming to EVA. It simplifies toolhead wiring significantly.