EVA-3D/eva-main #58: [EP] Change hole shape to make nut insertion into printed pieces easier

nesdnuma, 2021-08-17:
## Problem description
Insertion of nuts in printed pieces (especially small M3 nuts) can be a real pain, especially if there's not a flat surface around the hole. Sometimes one has to push very hard to put the nut in place. Moreover, when the holes are deeper than the nut height, it's hard to know when to stop pushing with the risk of breaking a printed part.

## Describe the solution you'd like

Make the holes in such a way that the insertion of the nuts are easier.

### Proposed solution

Change the shape of the holes. Only make the hole tight at the very end where the nut is supposed to stay. Otherwise make the upper part of the hole broader. Here are three different design profiles (#2 is my favourite).

Profile 1:

[image] [image]

Profile 2:

[image] [image]

Profile 3:

[image] [image]

nesdnuma, 2021-08-28:
No one's reading? :-(

EduardoOliveira, 2021-08-28:
Hi, have you tried pulling the nut into place by inserting a screw into the hole and thightening it into place?
That's what I usually do with a screw that's only the dept of the hole plus 3 or 4mm to grab the nut, after the nut is pulled into place i remove the screw and proceed with he assembly.

I know your suggestions simplifies the process but adds some work and provable collisions there the nut holes are close to other features.

pkucmus, 2021-08-28:
Thanks for the EP.
I honestly don't see the need for this, moreover, this is flawed as prone to alignment errors. One should not fit the nuts and after that thread the screws - we had a lot of issues with nut binding in the early days.
The proper way to seat a nut is with by ["backpulling"](https://main.eva-3d.page/printing/#handling-hex-nuts) this ensures that a nut will be driven into the socket on the angle the screw will want the nut to be.

nesdnuma, 2021-08-29:
For deep holes I still think it would be really useful and would not create alignment issues as the bottom of the hole is tight.  I have re-discovered backpulling on my side while building a V-Core Pro but the page you've linked to should be included in each Ratrig assembly documentation @pkucmus !

pkucmus, 2021-08-29:
It is on the one manual I made: https://ratrig.dozuki.com/Guide/00.+Preparations/78?lang=en#s477 
as for the rest, sure just let Rat Rig know :)

closed 2021-10-05 by stale[bot]