EVA-3D/eva-main #59: [BUG] E3DV6 not aligning with Titan extruder.

Keltbear, 2021-08-25:
## Describe the bug
A clear and concise description of what the bug is.

## Actual behavior
The hole on the thing does not match the hole on the other thing as I can't screw the two things together

## Expected behavior
I need the part to meet the other part but since the holes are not aligned I can't achieve that

## Pictures
[Photo](https://scontent.fsac1-1.fna.fbcdn.net/v/t39.30808-6/239427676_10225900459993710_8838118387369210191_n.jpg?_nc_cat=111&ccb=1-5&_nc_sid=825194&_nc_ohc=hid003mvOZsAX9K0VBV&_nc_ht=scontent.fsac1-1.fna&oh=d458c1a30d559f782a1199c50f055985&oe=6129E2D7)

## Device
 - Printer: [e.g. V-core ]
 - EVA version: [e.g. 2.1.0]
 - Mods:
   - Easy Mod 1.2.1

## Additional context
Add any other context about the problem here.

pkucmus, 2021-08-25:
Sorry, I can't figure out that "thing" which does not help me to figure out the problem not to mention a solution. 
Please edit the description and fill out what is being asked from the issue template.

EduardoOliveira, 2021-08-25:
@pkucmus I tagged you in a FB post that seems to be related to this issue

pkucmus, 2021-08-28:
This was resolved, the issue was a bad Titan Clone

closed 2021-08-28 by pkucmus