EVA-3D/eva-main #61: [EP]

paulokinho, 2021-08-28:
## Problem description
Support Biqu H2 Direct Drive Extruder as it's a very capable and lightweight extruder.

## Describe the solution you'd like

Biqu H2 it is very similar to Hemera extruder, so probably it's possible to the change the Hemera Front and Probe to fit it.

### Functional and non-functional requirements
- Front must match Biqu H2 holes
- Front must dimensions must match Biqu H2 dimensions
- Probe adapter must match Biqu H2 holes
- Probe adapter must  match Biqu H2 dimensions

### Proposed solution

Change Hemera front and probe adapter to match Biqu H2 holes and dimensions:
- Holes coordinates: 15mm between holes located at X 36.6mm, Y 8.9mm
- Dimensions: 35mm x 61.6mm x 35mm

https://3dwork.io/en/biqu-h2-extruder-review/

pkucmus, 2021-08-28:
This one I am going to reject. Compact extruders are not a good fit for EVA (or CoreXY printers in general) in the future I would rather separate those out do different projects and release those on a different release cycle. 
Maybe someone will like to pick that up as a contrib for now but it will not make it into official EVA.

paulokinho, 2021-08-28:
Ok, I see, just for curiosity, why a compact extruder is not a good fit for CoreXY printers?

EduardoOliveira, 2021-08-28:
They are usually wider than the other setups loosing X area, and since they are compact they are mounted in the from of the carriage as the other setups the motor sits on top of the carriage producing a more stable center of gravity

dlysi, 2021-09-20:
It;s a shame since Biqu H2 is only 61x35x35mm and 219g vs the Hemera which is 77x44x44 and 388g.