EVA-3D/eva-main #67: [Bug]: Eva 2.4.0 mounting holes for belt

Jegstrup, 2021-12-27:
### Discord username

Noobie4ever

### What happened?

hi 

when trying to mount belt on a sapphire plus the belt slider need to bee invertet otherwise the belt will be twistes any solution for this ? 

### EVA Version

2.4.0

### Which printer are you using?

Other

### Terms

- [X] I did seek help on Discord and found none or was instructed that this is indeed an EVA bug.

Jegstrup, 2021-12-27:
i did manage to solve it kind off :D it will work if i turn toolhead 180 degrees but prefer to have it like i should be

pkucmus, 2022-01-02:
So you just need tensioner blocks that would have the teeth on the other side?

Jegstrup, 2022-01-03:
Exactly. 😊

psusong, 2022-02-04:
[image]
only yes no no i searched all the stl

psusong, 2022-02-04:
Only tension_slider_9mm_belt_M5.stl
no opposite

pkucmus, 2022-05-21:
I'm closing this one as EVA 3 is here and I can't take on printer compatibility work, best to ask on Discord, a bunch of people there will be able to make adjustments

closed 2022-05-21 by pkucmus