EVA-3D/eva-main #68: [Bug]:  BMO Mounting Screws

zx4017555, 2021-12-27:
### Discord username

bobble#9830

### What happened?

Gonna make this quick and straightforward:

The Mounting Face([bmo_face](https://main.eva-3d.page/hotends/bmo/)) BOM says it requires M3X8 Screws for mounting 
[image]

However, in reality, it requires [M2.5X8](https://www.phaetus.com/dragonfly-bmo/) instead of M3X8 screws for mounting as per the documentation
[image]

A user suggested that the mounting bolts are included in the BOM, but the dragon has the M2.5X8 screws listed and it would be nice to have all of the necessary hardware listed. Second, the [CAD model](https://cad.onshape.com/documents/51ec5a95e190d928dabcea10/w/94576604d1279f8314466d44/e/02d805ddbc6466248e0837f4) shows only 2 out of the 6 Screws listed on the BOM, thus I'm supposing that the other 4 M3X8 screws are for mounting since you cant get 7mm screws.

Bcus of this, I bought the wrong mounting hardware and have to wait another week for new screws to arrive 😅 , wanted to suggest this to avoid future users from buying the wrong hardware

### EVA Version

2.3.0

### Which printer are you using?

Other

### Terms

- [X] I did seek help on Discord and found none or was instructed that this is indeed an EVA bug.

pkucmus, 2021-12-27:
The M3 x 8mm screws are for probe mounting, I had M2.5 screws in the BMO box - that's why those are not in the EVA BOM.
Please tell me if that is not the case for you.

zx4017555, 2021-12-27:
[image]
Then thats on my part, but from what I can tell, the Dragon has those 4 M2.5X8 on the bom, however, my BMO didn't come with the M2.5 Screws, from what I saw it should be included, Ill go ask for an replacement, thanks for the quick response!

pkucmus, 2021-12-27:
I need to check how Dragon ships nowadays, thanks!

closed 2021-12-27 by pkucmus