EVA-3D/eva-main #69: [Bug]: incorrect PTFE tube length for LGX + V6 combo

blalor, 2021-12-29:
### Discord username

yolo-dubstep#8033

### What happened?

Combined PTFE tube length (46.9mm) of LGX drive ([13.8mm](https://main.eva-3d.page/drives/lgx/#bill-of-materials)) and V6 hotend ([33.1mm](https://main.eva-3d.page/hotends/v6/#bill-of-materials)) appears to be too short.  When assembling my extruder I found that ~55mm is closer to correct, with the tube passing through the LGX coupler to make contact with the body, and fully inserted into the V6 heatsink until it stops.

### EVA Version

2.4.1

### Which printer are you using?

V-Core 3

### Terms

- [X] I did seek help on Discord and found none or was instructed that this is indeed an EVA bug.

blalor, 2021-12-29:
cut per spec:
[image]

what I'm using:
[image]

pkucmus, 2022-01-02:
Thanks, fixed in 2.4.2, the model for LGX did not assume us shoving the PTFE so deep, I measured my LGX and made changes to the model I hope it'll be correct now.

closed 2022-01-02 by pkucmus