EVA-3D/eva-main #73: Make EVA more modular[EP]

ManuelPeiro, 2022-01-10:
## Problem description
I am assembling the Ratrig v3 printer, and I have reached the point of mounting Eva, in fact I have already done so, my selection has been Bondtech Bmg GENUINE, with a pancake stepper and Mosquito clone.... And I am waiting Orbiter extruder. The problem I see is that when I get the new extruder or if I want another, I have to disassemble almost all EVA to be able to change it, losing the tension of the s
Belts, among other things ... However, if you switch from Hotend, it would be much easier

## Describe the solution you'd like
I propose do, top_bmg_mgn12, the part 
Short summary of what you want to happen.

### Functional and non-functional requirements
- functional requirement
- functional requirement
- non-functional requirement

### Proposed solution
The piece where Eva begins to be assembled, and which is the function of the extruder, divide it into 2. A piece with which Eva begins to be assembled, and that can be attached to (universal face or back corexy) and a second piece that holds the extruder and stepper motor

Later, if can, I add some hand drawings 

### Considered Alternatives
A clear and concise description of any alternative solutions or features you've considered.

### Additional context

The advantages, I think would be quite a lot, among which I see:
- if you change extruder you do not have to disassemble Eva. 
- You can adjust the Endstops easier. 
- You have a jam it would be easier to disassemble the extruder.

 I have to clarify that this platform, should be able to disassemble, with some screws, of that initial piece proposed

ManuelPeiro, 2022-01-10:
i have made hand drawing
[image]
s

pkucmus, 2022-01-10:
Thank you for this. I actually have thought about this but have a different approach where the top will be static and common for all drives. It's something I'd like to implement in the future version for sure.

ManuelPeiro, 2022-01-10:
If you need beta testers, count on me

closed 2022-02-16 by stale[bot]