EVA-3D/eva-main #76: [EP] Updated the Ender 5 Rear Plate to fit new size Tri Horn Ducts

Anon0ne, 2022-04-09:
## Problem description
Currently the Ender 5 backplate modification for cartesian style printer is the wrong size port for the Tri Horn Duct, I have started working on a new replacement for latest models, but having issues.

## Describe the solution you'd like

Updated flat back plate to save room on an Ender 5, which doesn't need belt tensioning or any channels for belt path.

### Functional and non-functional requirements
Just a fan and to fit the new ducts.

### Proposed solution

Updated flat back plate to save room on an Ender 5, which doesn't need belt tensioning or any channels for belt path.

### Considered Alternatives

Place the new style duct from back plate onto the existing ender 5 flat plate.

themacboy, 2022-04-13:
Not sure about your problem, check my PR. 

Thast my mod used in ender 5 plus

https://github.com/EVA-3D/contrib-extras/pull/47

pkucmus, 2022-05-21:
Closing as:
- printer compatibility is not something I can take on
- EVA 2.4

closed 2022-05-21 by pkucmus