EVA-3D/eva-main #77: Was support for the Dragon hotend removed?

atomicwrites, 2022-05-09:
I remember seeing the dragon in the list of hotends before, but it's gone now sometimes, or sometimes it's listed in the sidebar but the link 404s. I do see it in the v2.4.2 branch though. was this intentional?

EduardoOliveira, 2022-05-09:
Hi, official support was removed in version 3, in version 2.4 you should still be able to download directly from CAD

intelinc, 2022-05-11:
Bump. Just looking for this on 3.0.

rjorgenson, 2022-05-11:
This was mentioned in the announcement stream, they will not be supporting the Dragon hotend.

atomicwrites, 2022-05-11:
Not sure what's going on because I can't see it on the web UI, but I got this reply in an email:

> Hi, official support was removed in version 3, in version 2.4 you should still be able to download directly from CAD

rjorgenson, 2022-05-11:
> Not sure what's going on because I can't see it on the web UI, but I got this reply in an email:
> 
> > Hi, official support was removed in version 3, in version 2.4 you should still be able to download directly from CAD

The 2.4 version is here in the git repo https://github.com/EVA-3D/eva-main/tree/release/2.4.2/stls/hotends/dragon under the 2.4.2 release.

proap, 2022-05-12:
would the BMO mount work if we screw the BMO collar for the Dragon hotend?

pkucmus, 2022-05-20:
> Not sure what's going on because I can't see it on the web UI, but I got this reply in an email:
> 
> > Hi, official support was removed in version 3, in version 2.4 you should still be able to download directly from CAD

Who did you e-mail? EVA is just me and I did not wrote that :|

pkucmus, 2022-05-20:
The answer is here: https://main.eva-3d.page/FAQ/#dragon-support-clones-in-general 
I'll close this issue now :)

closed 2022-05-20 by pkucmus