EVA-3D/eva-main #80: EVA 3 - BMG vs Cable management Conflict

bubujoe, 2022-05-11:
BMG stepper motor is in conflict with the main cable management

[image]

pkucmus, 2022-05-21:
There is an alternative cable mount for it, I forgot to document it

pkucmus, 2022-07-09:
Fixed in 3.0.2

closed 2022-07-09 by pkucmus

david-antos, 2022-08-17:
The "fat" cable guide mount doesn't actually fix that completely: the BMG motor is about 2 mm too deep and conflicts even with the cable guide *mount*, not only with the cable guide ("the basket"). So something like, e.g. https://www.printables.com/model/203244-eva-3-modified-cable-guide-for-bmg-extruder is necessary (this solution works for 5015 fans, but conflicts with 4028, though).

olivbd, 2022-08-18:
> The "fat" cable guide mount doesn't actually fix that completely: the BMG motor is about 2 mm too deep and conflicts even with the cable guide _mount_, not only with the cable guide ("the basket"). [...]

I've the same issue with 4028 fan... You can check my work here: https://www.printables.com/model/261446-eva-3-modified-cable-guide-for-bmg-extruder-with-4