EVA-3D/eva-main #83: Bl-touch mount

duniu9, 2022-05-17:
The Bl-touch mount is too thick and collides with the sensor. There is 8mm space for the bl-touch, the mount is 9.5mm thick.
[image]

duniu9, 2022-05-17:
Another problem is the sensor collision with the shroud
[image]

mikejwaters, 2022-05-18:
I also had the same problem. I shaved away the excess after printing to put the BLtouch on it's mount.

pkucmus, 2022-05-21:
The shroud compatibility is something I'll think about more, will likely produce many different shrouds.
As for the mount, please test this: https://cad.onshape.com/documents/cb8bd513b980b411d50cd513/w/6f7fd7828c4d7cb1c0c853da/e/dbf188e21e46a24ca801cc78

mpx1986, 2022-05-21:
the next Problem ist the fan shroud

mbiedak, 2022-05-24:
New version is correct in height, but I am unable to fit it with cables attached. You can see that on the picture above

mbiedak, 2022-05-25:
@duniu9 can you share with me your updated file?

duniu9, 2022-05-26:
> @duniu9 can you share with me your updated file?

[Bl-touch fix.zip](https://github.com/EVA-3D/eva-main/files/8777071/Bl-touch.fix.zip)

pkucmus, 2022-05-26:
BL-Touch is tricky as it has many versions and revisions, some say it works for some it does not, can you post real life pictures here for me to better understand what is going on?

sladewexo, 2022-05-30:
Though the bl-touch fitted for me, it was sitting to low on my V-Core 3 500.  I altered @pkucmus design a bit to move it higher up.  For those interested, I've included the STL and STEP file here
[bltouch.zip](https://github.com/EVA-3D/eva-main/files/8796484/bltouch.zip)
.

mikejwaters, 2022-06-01:
> BL-Touch is tricky as it has many versions and revisions, some say it works for some it does not, can you post real life pictures here for me to better understand what is going on?

@pkucmus I know it's inelegant, but would having a printable shim help with BL-Touch compatibility? Just a thought.

pkucmus, 2022-06-12:
I made a bunch of improvements to the BL-Touch mount, please tell me how it works, EVA 3.0 was just released: https://main.eva-3d.page/heat_insert/probes/bl_touch/

closed 2022-06-12 by pkucmus

b33k-WA, 2022-06-13:
I really struggled to get the nuts into the recesses, perhaps a little more tollerance, I ended up having to print the piece 3x because of bending the thin arm when trying to pull the nut into place and another when it spun.  I ended up printing at a high infill and setting the nut like a brass insert (sorry I know you hate those).  Once I got it on the machine I encountered issues with not having any adjustment available and felt it was too close to run. 

[image]
![IMG_1334](https://user-images.githubusercontent.com
[image]
/95461285/173274287-9dac702b-dccb-49d5-bbc8-492a2a12534c.jpeg)