EVA-3D/eva-main #85: Request. E3d Revo Voron mount

midicdj1000, 2022-05-25:
Hi, Will an EVA 3 mount for the E3d Revo Voron be in the future  ? It's not the same as the BMO.
https://e3d-online.com/blogs/news/revo-voron-available-now
I am just printing a BMO set at the moment in PC CF and it looks fantastic

midicdj1000, 2022-05-25:
HI, I just saw that I can download the step file so I can edit in 123design my normal cad software. if I make my own parts for my diy printer what should I do so others can use ? I just made the top part with limit switch to fit my mgn12H part instead of the mgn12c for example.

pkucmus, 2022-05-30:
I just purchased a Revo Voron with donation money, I'll make the mount for it soon

midicdj1000, 2022-05-30:
Thanks, I am almost finished with this EVA3 for a dragonfly. I printed in Prsament PC CF and Esun silver PETG
[image]
[image]
[image]
[image]

pkucmus, 2022-06-12:
Released: https://main.eva-3d.page/heat_insert/hotend/revo_voron/

closed 2022-06-12 by pkucmus

midicdj1000, 2022-06-13:
Thank you. What is the link to send you a small donation ?