EVA-3D/eva-main #87: [EP] EVA 3 E3D Revo V6 Support

githubcura35, 2022-06-14:
EVA 3 now supports the E3D Revo Voron. Are there any plans to support the E3D Revo Six? (https://e3d-online.com/products/revo-six) Any plans to support V6 based hotends at all?

pkucmus, 2022-07-01:
This would also mean support for V6, right? Not sure about the timing but something I'd like to address

huzargarden, 2022-08-27:
Definitely, something that I'll be looking for. I like the EVA design and I'm very happy with my V6 Hot End.

a937983423, 2022-09-21:
I would like to ask why the most popular E3D_ V6 does not support

fj-guy, 2022-10-13:
The old e3d v6 served me well for quite awhile.  Since the Revo 6 was a direct drop-in replacement I switched to that. And I would imaginge a lot of V6 users did the same.  Not all of us care to print at 300mm/s with crazy accel values.  I recently upgraded to the V-CORE-3.1 but had retained the older 2.4 carriage because I didn't want to upgrade to a HF hotend just yet.  Luckily someone did a V6 mount for EVA 3.  Do your survey on REVO 6 users instead of V6 users.

MechanizedMedic, 2022-10-25:
I just got done drawing up a [to-spec remix](https://cad.onshape.com/documents/aa6b87b04c53d0436ebc5b95/w/52b723d56fd8bb8eb734a789/e/abb8bc791c22639c7a986955?configuration=List_G0JwUrcDLEaXCX%3DBL_Touch%3BList_Ojsu1VJQPwv42D%3DBMG%3BList_S9vpzEZQSzFGKh%3DVolcano%3BList_SYMLdq345kAX9B%3DVolcano%3BList_durAPyfJdcZirJ%3D_5015%3BList_vN2ntClvCm9on0%3DNone&renderMode=0&rightPanel=configPanel&uiState=63578f586b23cf60326f8b8e) that is compatible with Volcano/V6 (if push fitting is removed). It should allow the Volcano and v6 to use the same hotend block with the standard cooling duct.

I also made a cable mount for the BMG extruder but it is only compatible with the single 50mm blower for now.

Feedback and collaboration are welcome! :)

githubcura35, 2022-10-25:
@MechanizedMedic Thanks a lot. I am unable to export to STL. Can you grand permissions in Onshape?

MechanizedMedic, 2022-10-25:
@githubcura35, yep sorry about that. Should be good now.

githubcura35, 2022-10-27:
@MechanizedMedic Thanks a lot. I will check it out but feedback will take some time.

githubcura35, 2022-10-31:
@MechanizedMedic The V6 fits perfectly but I am having some trouble with the cooling. The duct from the default bottom reaches lower than the tip of the nozzle
[image]

githubcura35, 2022-11-01:
@MechanizedMedic There is no way to move "hotend_volcano" further down as the zip tie holder collides with the "universal_front".
[image]

MechanizedMedic, 2022-11-02:
Okay, thanks for testing. I'll fix it up later this evening and repost.

olivbd, 2022-11-02:
Why not use this Printables model? https://www.printables.com/model/201093-eva-30-e3dv6-revo-revo-micro-hotends-mount

I printed and used it for several months without any problem...

MechanizedMedic, 2022-11-02:
@githubcura35, I updated the design this morning to move the hotend down 7.5mm. This should put the v6 in the correct spot for the standard duct. The Volcano hotend will now need the Volcano duct.

TekuSP, 2023-06-30:
> Why not use this Printables model? https://www.printables.com/model/201093-eva-30-e3dv6-revo-revo-micro-hotends-mount
> 
> I printed and used it for several months without any problem...

That is REVO V6 and not V6 standard