EVA-3D/eva-main #90: Orbiter + Mosquito = Gap!

Tylersuard, 2022-06-24:
Currently the EVA combo for Orbiter and Mosquito introduces a gap where the filament is just hanging in the breeze for about 15 mm.  The Orbiter is specially designed to constrain flexible filament so it has nowhere to go.  Adding a gap below it completely defeats the purpose!

it would be nice if we could get a version with a dorect connection from the orbiter to the mosquito.

Yako000, 2022-06-24:
You are suppposed to add a piece of bowden tube between the extruder and the hotend. It should tell you how long it needs to be in the list of required hardware.

pkucmus, 2022-07-01:
Yeah, I think @Yako000 nailed what might be wrong here.
Please see https://main.eva-3d.page/getting_started/#ptfe-lenghts and:
[image]

closed 2022-07-01 by pkucmus