EVA-3D/eva-main #93: EVA3 - BLTouch `horizontal_move_z can't be less than probe's z_offset`

DrMullings, 2022-07-02:
When using the EVA3.0 or EVA 3.01 version of the BL Touch Holder I have Z offsets from 14 to 16 and therefore neither Z Tilt, nor the heightmap calibration works without adapting horizontal_move_z (default are 5 and 12) in the printer.cfg
This adaption only seems to be a workaround, as it was not necessary in EVA 2.4

joaobarros, 2022-07-02:
This looks like an issue to be open in RATOS, not here

DrMullings, 2022-07-02:
> This looks like an issue to be open in RATOS, not here

This was introduced by switching from EVA 2.4 to 3.0
That is the reason why I opened the bug here and not in RatOS

pkucmus, 2022-07-05:
#91 should fix this or it's indeed something for RatOS to address. 
@miklschmidt is this anything you can help with?

pkucmus, 2022-07-07:
I'm closing this as not EVA related. If you need further help I suggest Discord :)

closed 2022-07-07 by pkucmus