EVA-3D/eva-main #101: Biqu H2 extruder/hotend

Twistedsocal, 2022-08-02:
Any chance of getting support for the biqu H2? I just got the H2o which is water cooled and really want to be able to use it with this setup. Thanks. I'm happy to Paypal for the effort.

pkucmus, 2022-08-03:
Nope sorry, my stance did not change since this comment: https://github.com/EVA-3D/eva-main/issues/61#issuecomment-907641956

closed 2022-08-03 by pkucmus

pkucmus, 2022-09-30:
For context, EVA had Hemera (I'll compare H2 to Hemera here as the reasoning is the same for both) up to I think 2.3 or some version like that, people did not care for it more many reasons (3.1% of the users were on Hemera). The same story was with the Aero.

We called them compact or "front heavy" extruders. With the "normal" setup you always have the motor (heaviest part) on top and very close to the MGN carriage. Where those compact extruders have their center of mass shifted to the front where the weight can sag the whole assembly. They also stole print volume, so the math is:

+ whatever there is that is good about those (I honestly do not see the appeal)
? a shorter filament path, but that was proven to have no or negligible impact on everything
- close to no interest from the community
- tons of extra work
(and cons that are more universal in terms of CoreXY)
- need more space in the front of the carriage (i.e. for EVA compatibility the printer would need to reserve much more XY space for EVA)
- the center of gravity is further away from the anchor point (MGN carriage)
- probes always had to be in weird places

Additionally about the weight gain: H2 is 219g, the rather huge LGX is 218.6g, LGX Lite is 48g add a typical hotend ~ 41g (Mosquito) - unless H2 has more push force (which I have no data for but I seriously doubt that) I see no reason to use H2 on any printer (if money is no issue).

If you'd want to discuss this further we could take it to Discord, comments here are not the best for this kind of discussion.

dlysi, 2022-09-30:
@pkucmus Now with H2 V2S Lite and and Hs V2S revo i believe you should rethink this.
H2 V2S Lite is 175g with 45mm3/s flow volume 
and 
H2V2S Revo is 198g

https://biqu.equipment/collections/h2-series-extruder/products/h2-v2s-lite
https://biqu.equipment/collections/h2-series-extruder/products/biqu-h2-v2s-revo-extruder

pkucmus, 2022-09-30:
@dlysi still does not compare to a LGX Lite and any hotend able to push that much filament, but even if it would compare it only addresses one argument. Not sure what there is to rethink.

If I had the time to add support for it it would be a standalone EVA Compatible-ish project that would live outside of EVA's release cycle - it would probably involve a special front and bottom. Maybe someone would like to pick that up? I have high-demand hotend support lined up and I'm super late on that.

dlysi, 2022-09-30:
@pkucmus i agree with this... "it would be a standalone EVA Compatible-ish project that would live outside of EVA's release cycle"

Twistedsocal, 2022-09-30:
Ok, thanks for the clarification. To be clear, the H2 is not the hemera,
not anything like the hemera, the hemera is larger, heavier and less
capable across the board, the H2 which I got because of the ability to turn
retraction settings down to 0.2mm and to do so rapidly means no artifacts
in the print. The ability to reach 500c in the same setup so long as you
were to swap the heartbreak out for the titanium version which I have and
the extremely short filament path great for flexible filament. Plus a one
stop shop plug it in turn it on and run it type of deal and now they have
improved it even more. Idk just wanted to fill you in, still thanks for the
info, too bad you are missing out on this pretty capable awesome product.