EVA-3D/eva-main #102: V6 hotend for EVA3

ZeyHex, 2022-08-03:
E3D V6 is the most common platform at the moment, which is still a relevant and productive solution, and I find it strange that V6 heatsink support was removed from EVA 3. I suggest returning it. To do this, I, in the image and likeness of the original solutions, modeled a technological bed for the V6. Model attached there: https://github.com/ZeyHex/eva3_hotend_v6

[image]

I ask you not to condemn the applied solution for fixing the radiator in the glass bed immediately with a bayonet. The radiator is mounted in it once and no longer requires disassembly. If you choose the right distance between the holes, the screws firmly and securely fix it both from falling out and from turning. And besides this, the rigidity and protection against inertia of the structure is given by the emphasis of the walls of the radiator in the body of the bed from the inside.

[image]

You can take the model that I attached as a basis and modify it if something is wrong in it.

pkucmus, 2022-08-03:
Hi, thanks. This is a duplicate of #87 closing - V6 support will happen in that ticket.
I would like to address the "most common platform" bit. According to the EVA survey taken more than a year ago, V6 had 11.5% of the usage I can only imagine that number being lower now.
I don't understand the second part with the glass bed and bayonet, can you explain? I also don't have Fusion360 so I can't open the attached file.

closed 2022-08-03 by pkucmus

ZeyHex, 2022-08-03:
added step
https://github.com/ZeyHex/eva3_hotend_v6/blob/main/EVA3%20v6%20hotend.step

elcojacobs, 2022-08-04:
https://www.printables.com/model/201093-eva-30-e3dv6-revo-revo-micro-hotends-mount/files

This one works well already.