EVA-3D/eva-main #108: Inferoir cooling for mosquito throat radiator

AlexTOOT, 2022-10-25:
Hi, I'm using EVA3 on a corexy machine with mosquito. 

The back of mosquit is almost closed by the extruded aluminum chassis, causing the radiator cooling fan to stall creating tons of noise and insufficient cooling for throat. (Not part cooling, trihorn works perfect)

with that happening , when long time printing with high speed and high temp, sometimes the softened plastic goes up the throat and almost get into extruder causing severe clogs. Any tips? 

Thanks for the stunning design BTW.

pkucmus, 2022-10-31:
Hi, thanks. What do you refer to when speaking of "extruded aluminum chassis"? The X gantry 2020? My first guess would be a problem with the fan itself. I'm currently running a Mosquito HF on my main VC3.1 with a default 4010 GDS fan and have no issues, neither with clogs or noise.

AlexTOOT, 2022-11-04:
> 

yes, X gantry. forgot its name in English. 

I tested several sunon 4010s, a sunon 4020 and a delta 4028. They all have the same issue but varies a little, the 4010 is easier to recovery from the stalling state by blowing into it from the front. I believe that breaks the vortex or some sort,  force it to re-stabilize. 
The other two are a little bit harder to recover but they eventually do recover.  And when they get stable, they stay that way till shutdown.
 I'm not quite sure what is going on, so... maybe i'll do more test and let you know.

TekuSP, 2022-11-13:
If you have board supporting 12V, try using Noctua 4010. On paper should have higher airflow than most sunons, and mainly a lot quieter, so it can run at higher RPMs.