EVA-3D/eva-main #109: BLTouch with Mosquito on same level

Seven4ninE, 2022-11-18:
I have a BLTouch + Mosquito with NoozleX and LGX Mini.The BLtouch is at the same Level as the Noozle.
[image]

olivbd, 2022-11-21:
Do you have a guenine BLTouch or a clone?

Mine is a 3D-Touch (clone) and I've same issue with BMO hotend... so I designed a 2mm raised version of the bracket.

You can find it here https://www.printables.com/model/322057-eva-3-modified-bl-touch-mount

Seven4ninE, 2022-11-21:
Hi,mine is an original and i made now a Version That is 1mm higher. Thank you for your reply .RegardsHarryVon meinem iPhone gesendet