EVA-3D/eva-main #114: Incorrect toothed belt profile at the slide tensioner

ZeyHex, 2023-02-26:
It's about this detail:
[image]

At the moment, the profile of the channel of the toothed belt is as follows:
[image]

However, a typical toothed belt profile is as follows:
[image]

Even in the model, this movable belt tensioner does not inspire confidence in its reliability, but in reality it is not uncommon for 1-2 belt teeth to pop out of the tensioner when the belt is tensioned. And sometimes, with sufficient jerks and accelerations, the belt can completely pull out of the tensioner.

For myself, I solved the problem by narrowing this place by 0.2 mm, and now the part behaves quite reliably, however, I think that the model in the project needs to be corrected in such a way that it ensures that the belt is securely held in any conditions.
[image]

ZeyHex, 2023-07-19:
This is an awesome solution! How did I not think of it myself .. Super!

I also switched to a 9mm strap on the print head and that also helped it stay on the slider.

ricki158, 2025-02-10:
Have you STL reviewed with this solution?