EVA-3D/eva-main #117: Fan duct blows crookedly

ZeyHex, 2023-06-04:
Hello!
I did not have enough airflow in the configuration with 5015 and I installed server 4028. It turned out that already at half power the air blows beyond the nozzle and forward, and not down on the part. As it shown on the picture.
[image]
I think we need to extend the air channels a little closer to the nozzle to straighten the flow more, and make them smaller at the exit so that the flow gets where it needs to be with more speed and pressure.
For example, as shown in the pictures below.
[image]
[image]
I also understand that some people still use outdated oversized heaters and this option will not work for them. In this case, the flow can be straightened by bending the channel more, as shown in the figure.
[image]

In the current implementation, powerful airflow through such channels is simply impossible, they blow absolutely in the wrong direction when the fan accelerates to large flows.

dstrelnikov, 2023-06-09:
I just replaced 2x5015 with a powerful 4028 fan and can confirm the issue. Even at 50% the airflow shifts to the front from the nozzle.

@ZeyHex, please let me know if you get some improvement with an adjusted duct.

MudiStar, 2024-04-20:
I don't know if this is still a topic for you, but try the tri horn
The airflow seems to fit 
https://main.eva-3d.page/heat_insert/bottoms/tri_horn/#description

ZeyHex, 2024-04-20:
> I don't know if this is still a topic for you, but try the tri horn The airflow seems to fit https://main.eva-3d.page/heat_insert/bottoms/tri_horn/#description

What's the difference? It also blows crookedly.
Apparently, EVA is dead, now RatRig make their own head and airflow.